ControlSystemStudio/phoebus #3323: Display builder editor feature - use display grid size when moving widget

UrbanBobek, 2025-03-11:
Hi all,

when we are editng screens we noticed that is very handy to move widgets around using the arrow keys. Currently, holding the *ctrl* key moves the widget in 10px increments, but recently we started using a grid size which is not a multiple of 10 and we noticed that it would be very practical that when holding the *ctrl* key the widget position is incremented for the grid size value.

We already implemented this feature on our fork of Phoebus. I can open a PR for this feature if it is something that you would like to see in the display builder.

Best,
Urban

kasemir, 2025-03-11:
Sounds good, please go ahead with PR

kasemir, 2025-03-11:
While you're at it: Holding SHIFT and then using the cursor keys will resize the widget.
You could make it resize in grid steps when holding both SHIFT and CTRL

UrbanBobek, 2025-03-11:
Great, I will open a PR. 

Oh right, thanks for pointing this out. We also implemented this, but I forgot to mention it.

kasemir, 2025-04-01:
Done, thanks!

closed 2025-04-01 by kasemir