ControlSystemStudio/phoebus #3340: Failure to convert some OPI actions to BOB using the converter

rjwills28, 2025-03-25:
I have found a few bugs relating to the conversion of OPI files to BOB format. Note that all 3 of those listed below worked in version 4.7.3. It is only in the version I've built from GitHub master where it fails indicating that these bugs have been introduced since the last release.

**ISSUES**

1. This relates to the actions that are attached to say an ActionButton or any other widget. If you have an OPI widget with an action to **EXECUTE_CMD** or **EXECUTE_PYTHONSCRIPT**/**EXECUTE_JAVASCRIPT** they fail to be parsed and so are removed. I see the following error:

    ```
    2025-03-21 09:37:15 SEVERE [org.csstudio.display.builder.model] Error reading widget Widget '' (action_button) property <actions>, line 91
    java.lang.RuntimeException: No action implementation matching type 'EXECUTE_CMD' found.

    2025-03-21 09:37:15 SEVERE [org.csstudio.display.builder.model] Error reading widget Widget '' (action_button) property <actions>, line 144
    java.lang.RuntimeException: No action implementation matching type 'EXECUTE_PYTHONSCRIPT' found.

    2025-03-21 09:37:15 SEVERE [org.csstudio.display.builder.model] Error reading widget Widget '' (action_button) property <actions>, line 198
    java.lang.RuntimeException: No action implementation matching type 'EXECUTE_JAVASCRIPT' found.
    ```

    The problem here I think relates to the comparison of the 'type' attribute. I needed to make one change to the ExecuteCommandAction.java and 2 changes to the ExecuteScriptAction.java to get these to parse correctly.

2. Once I'd fixed the above I no longer saw the exceptions and could open the converted BOB file to see that the actions were indeed there, however in the case of an embedded EXECUTE_PYTHONSCRIPT/EXECUTE_JAVASCRIPT the actual EmbeddedScript was missing again. The error it throw is rather obscure but digging down I found this is because there is also a bug in the `writeToXML()` method meaning it fails to detect that these should be embedded scripts. 

    ```
   2025-03-24 10:03:20 WARNING [org.csstudio.display.builder.runtime] Script for action failed to compile
    Display '/home/user/CSS/opi_actions.bob', Widget 'Action Button_3' (action_button), Execute Script
     java.io.FileNotFoundException: /home/user/CSS (Is a directory)
	at java.base/java.io.FileInputStream.open0(Native Method)
    ```

3. Finally, when using the org.csstudio.display.builder.model.Converter tool I see an exception that it fails to close the file. This is because the `close()` is being called twice, once explictly in the method and the other because a try-use-resources block has been used meaning that it performs an autoclose when the resources has been finished with. Solution is to remove the explicit `close()` in the `convert()` method as this is purposely not needed now.

    ```
    java.io.IOException: Failed to close XML
	at org.csstudio.display.builder.model.persist.ModelWriter.close(ModelWriter.java:230)
	at org.csstudio.display.builder.model.Converter.convert(Converter.java:179)
	at org.csstudio.display.builder.model.Converter.convert(Converter.java:219)
	at org.csstudio.display.builder.model.Converter.launchConversion(Converter.java:247)
	at org.csstudio.display.builder.model.Converter.main(Converter.java:263)
    ```

rjwills28, 2025-03-25:
I have fixes for these 3 issues in my forked branch: https://github.com/rjwills28/phoebus/tree/opi_converter_fix and can open a PR for review. I have also added 2 unit tests to test this functionality so that we may catch any regression going forward.

georgweiss, 2025-03-25:
This is most probably regression caused by a rework of the "action" feature where we've introduced a modular approach. The API now allows for detection of actions exposed over SPI.

rjwills28, 2025-03-25:
PR merged - thank you for the quick response getting this reviewed and merged.

closed 2025-03-25 by rjwills28