ControlSystemStudio/phoebus #3346: Text message support?

kasemir, 2025-04-01:
The alarm server supports "automated actions" to send emails.
For us, a key use case was based on mail addresses like `mailto:5551234567@txt.att.net` which the cell phone provider then turns into text messages to that cell phone number.
Details differ for each cell phone provider, and such services are being closed down, 
https://www.att.com/support/article/wireless/KM1061254/

There are commercial offerings, subscriptions to web services for submitting text messages, but no obvious choice for a free one.

Unclear what to do.

georgweiss, 2025-04-02:
At ESS we have a service listening for alarm Kafka messages, which trigger push messages to IOS and Android clients. In those apps users can select which topics to "subscribe" to. 

This solution is however not targeting alarms specifically. Any system can post messages to the service for dispatch to IOS and Android clients. We use it also for the logbook, i.e. when new log entries are created, clients will be notified and user can see log entry digest.

Code is open source, we're happy to share. All free, except charges imposed by Apple and Google for developer accounts.

georgweiss, 2025-04-02:
Developer accounts is all it takes.
Apple and Google push mechanism use device specific tokens, but that is managed by the platforms. On the service one needs to keep track of the tokens in order to push messages, so clients will need to register the token with the service. All this is implemented in the service and the clients we manage. But of course the tokens are issue by Apple and Google and bound to the respective apps.

Oh, one more thing. The "MAX IV" synchrotron next door is also using this setup.

kasemir, 2025-04-02:
I assume you have several network layers between the control system and the world.
For example, within our controls network, the alarm server has access to an smtp mail server, which then forwards the emails to another mail server and finally to the world.
What's your pipeline from an alarm server on the controls network to the Apple/Google web services?

georgweiss, 2025-04-02:
I would need to check that, but for the time being I will assume the service is allowed to consume Kafka messages from the controls network, and that it is located in a zone with internet access.

georgweiss, 2025-04-03:
So the full story is that our notification service is essentially a message broker. It runs on a network with internet access as it needs to post to Apple/Google push services. Internally it provides an API accepting requests to push messages. Another service acts as Kafka client, repackaging messages to fit the general purpose API and sends them off to the notification service. Yet another service is called by the Olog service whenever a new log entry is created, and this in turn does the same trick.

We've setup Kafka on the controls network to send messages to subscribers across network boundaries, but not to accept producer messages from outside the controls network.

georgweiss, 2025-04-03:
Notification is based only on the tokens associated with the Apple and Google push message framework. No phone numbers. On the client side app lists the available service ids which the user may subscribe to. On the service we maintain a list of service ids for which a user (=push token) has subscribed. So push messages are dispatched based on what user selects in the app (which lists all available service ids).

georgweiss, 2025-04-03:
It's a custom app, hence my comment on the need for developer accounts with Apple and Google.
IOS and Android app source code is available.