ControlSystemStudio/phoebus #3356: Problem with Save and restore Service and ChannelAccess protocol

sk359, 2025-04-10:
Since a few days saving a snapshot with the save and restore service does not work anymore. It is used with simulated PVs using the ChannelAccess protocol (created with Pythons caproto library). It throws the exception

```
save-and-restore-1          | Apr 10, 2025 10:58:32 AM org.phoebus.pv.PVPool <clinit>
save-and-restore-1          | INFO: Default PV type ca://
save-and-restore-1          | Using default EPICS protocol: ca

...

2025-04-10 10:50:02.759  WARN 1 --- [ool-2-thread-10] o.p.saveandrestore.util.SnapshotUtil     : Failed to read PV 'DEV_HA:IN2:OVN1-V1Set'
save-and-restore-1          | 
save-and-restore-1          | java.lang.Exception: DEV_HA:IN2:OVN1-V1Set has unknown PV type 'ca'
save-and-restore-1          |   at org.phoebus.pv.PVPool.getPV(PVPool.java:211) ~[core-pv-4.7.4-SNAPSHOT.jar!/:na]
save-and-restore-1          |   at org.phoebus.saveandrestore.util.SnapshotUtil.lambda$takeSnapshot$1(SnapshotUtil.java:127) ~[save-and-restore-util-4.7.4-SNAPSHOT.jar!/:na]
save-and-restore-1          |   at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[na:na]
save-and-restore-1          |   at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) ~[na:na]
save-and-restore-1          |   at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) ~[na:na]
save-and-restore-1          |   at java.base/java.lang.Thread.run(Thread.java:840) ~[na:na]

```
This happens with all IOCs and devices. At the start of January this still worked. since then the provided docker-compose file has more configuration concerning the pvAccess protocol, maybe this is the problem. I tried some to make it clear to use ChannelAccess, without any success. Any ideas what the problem is?

georgweiss, 2025-04-10:
PVAccess from save&restore inside Docker is problematic and requires tricks to work.

As for your stack trace... Seems like the save&restore service was built without core-pv-ca. How do you build the service?

sk359, 2025-04-10:
Thank you for you response. The jar file is build with

```
mvn --batch-mode install -DskipTests
```

sk359, 2025-04-10:
As far as I can see the pom.xml file is not modified. The repository (branch master) was cloned and than a specific commit (last time: 6ffdefbff9143a1ec06a0558fe05a7b1e2e610ab) was checked out. Within this code the mvn command was executed.

sk359, 2025-04-11:
You were right. I tried several commits and the first one that failed was f6c574af60b6d57104899f96accf13d2bc48308e from Februray 19th, where the save and restore pom.xml was changed. The changes where undone with commit 3df09850bb938c40d2b65db87555856b1bffdc7a from March 5th, with that the error disappears. Thank you!

closed 2025-04-11 by sk359