ControlSystemStudio/phoebus #3366: Feature: Configurable Runtime Zoom Factor

UrbanBobek, 2025-04-14:
Hi all,

we implemeted a way to set a zoom factor for the runtime screens in the `settings.ini` file. 

The motivation behind this was to have a way to make GUI sizes look consistent across different monitor sizes and resolutions, but it can also serve as a way for operators to costumize GUI size for their preference. 

Is this someting you would like to see in the master branch? 

Cheers,
Urban

georgweiss, 2025-04-14:
@UrbanBobek, are you able to provide an example?

UrbanBobek, 2025-04-15:
You can specify the zoom factor in the settings file and all screens that are opened have the default 100% zoom multiplied by this factor:
```
##################################################
# Package org.csstudio.display.builder.runtime.app
##################################################

org.csstudio.display.builder.runtime.app/default_zoom_factor=1.23
```

[image]

I can open a PR if that will help you with understand the functionalty better.

georgweiss, 2025-04-15:
So this would in practice override the default zoom factor of 1.0.

I cannot see why this should not be supported.

closed 2025-04-22 by UrbanBobek