ControlSystemStudio/phoebus #3367: Future of take-snapshot endpoint in Save and Restore service

sk359, 2025-04-15:
This is rather a question than an issue: until recently (January?) the `take-snapshot` endpoint was part of the API documentation of the Save and Restore service, but not anymore. The endpoints in the controller are decorated with `@SuppressWarnings("unused")` (see [file](https://github.com/ControlSystemStudio/phoebus/blob/master/services/save-and-restore/src/main/java/org/phoebus/service/saveandrestore/web/controllers/TakeSnapshotController.java)). The endpoint still works. In the current documentation there is only  `PUT /snapshot` as an alternative, but the list of PVs with its current values has to be build manually as it seems. It is very handy to have this done automatically by using a configuration node. Is there a plan to remove the `take-snapshot` endpoint in the future?

georgweiss, 2025-04-15:
Will check if documentation needs update.

No, there is no plan to remove the  ```take-snapshot``` endpoint.

georgweiss, 2025-04-16:
@sk359, the reason endpoint methods are annotated ```@SuppressWarnings("unused")``` is that these are implicitly used by the Spring Boot framework. There is no code in the project calling these methods, and with that in mind I want to suppress warnings (improves Sonarqube metrics...).

sk359, 2025-04-17:
Ok, thank you for your answer!

closed 2025-04-23 by sk359