ControlSystemStudio/phoebus #3372: Channelfinder PV Completion in Phoebus

eddybl, 2025-04-16:
Hey,

as far as I know you can use the Probe of Phoebus and use "auto completion" / lookup of PVs which are present in ChannelFinder.

We have Phoebus and ChannelFinder running, but I am not sure how to get that part working? Is this a Phoebus configuration? CHannelfinder? Probe Hotkeys?

shroffk, 2025-04-16:
as long as you are using the common product or ensure that thge `app-utility-channel` is part of your product it should work out of the box.

You should verify if the following preference is properly set
`org.phoebus.channelfinder/serviceURL=http://localhost:8080/ChannelFinder`

eddybl, 2025-04-16:
Thanks! It seems like we were missing the `/ChannelFinder` part of the ChannelFinder service URL

closed 2025-04-16 by eddybl