ControlSystemStudio/phoebus #3378: Phoebus save-and-restore service embedded LDAP and hashed passwords

minijackson, 2025-04-22:
Hello, I'm trying to set up the Phoebus save-and-restore service, with the embedded LDAP authentication implementation, and I'm trying to input hashed passwords, but can't figure out how to make it work.

I know in OpenLDAP, you can specify password like this:

```ldif
dn: uid=customAdmin,ou=Group,dc=sar,dc=local
uid: customAdmin
# ...
# Either unencrypted
userPassword: customAdminPass
# or salted SHA
userPassword: {SSHA}GuzFEUlPyBZ7ZbPKn2GdclD8qd5duw+x
# or using crypt with bcrypt
userPassword: {CRYPT}$2b$05$zUX77ddLbxKLvPxTbsLot.Zq6m7Z2.q/p.qpnLyReQmknMRxWAr6y
```

But any attempt at using hashed password fails, and Phoebus save-and-restore respond with 401 Unauthorized.

Since we use Git to track configuration files, and deploy them with Nix which puts them in a world-readable location, I'm interested in not having them in plain text.

Have you got a way to do this? Thanks a lot!

georgweiss, 2025-04-22:
@minijackson, I have not had any use case to test encrypted password for embedded LDAP. Are you able to sign in if you put unencrypted password in the config file?

minijackson, 2025-04-23:
@georgweiss yes, logging-in works with an unencrypted password. I'm also using a custom user name and password to be sure that the LDIF file is taken into account

georgweiss, 2025-04-24:
@minijackson, what if you remove ```{CRYPT}``` from the bcrypt password?

minijackson, 2025-04-24:
@georgweiss I've tried several "crypt" password, without `{CRYPT}` as you suggested, including bcrypt, SHA512, and yescrypt, but none of them worked, unfortunately

minijackson, 2025-04-24:
Yes, I'm commenting other lines, so that there's only one entry

georgweiss, 2025-04-24:
Sorry, I do not have an answer off the top of my head. Would need to investigate. 

About encrypted password: how is it encrypted? Asking as I assume a key would be needed to decrypt it?

georgweiss, 2025-04-25:
@minijackson, I think I have found a way to solve this using bcrypt. However, since ldap related code is used both for embedded and remote ldap, I would need to restructure a bit. And of course verify that remote ldap still works :-)

How urgent is this?

minijackson, 2025-04-25:
@georgweiss thanks a lot! I don't think it's really urgent, it should still be possible to go around the issue by using a separate OpenLDAP server.

georgweiss, 2025-04-25:
OK, I'll put this on my backlog.

georgweiss, 2025-04-29:
@minijackson, migration to Java 21 and Spring Boot 3.4.4 depends on a common effort of verification, see #3377. Just saying ;-)...

minijackson, 2025-04-30:
@georgweiss oh I can certainly help with that! I'm on vacation currently, so I can only help in around 2 weeks.