ControlSystemStudio/phoebus #3379: Warning messages in log "xx is not a URL" every time a display is opened

rjwills28, 2025-04-23:
Every time a display is open, whether that is from the file browser or from an Open Display action, you see a log warning:

```
WARNING [org.csstudio.display.builder.model] URL /home/user/bob/test.bob is not a URL
```

I can see that this was added in the case that you are trying to open a webpage (see [commit](https://github.com/ControlSystemStudio/phoebus/commit/46a3908a021317cb0295e9b03bec8fd428c6dcb3)), but the same method is called to open any resource and so I'm not sure this message should be a warning as it doesn't seem appropriate when trying to open a file?

kasemir, 2025-04-23:
See also #3274. So you get the superfluous warning, we had to edit our display links to remove older `*.opi` references. Not sure who's happy with the current implementation,

DenSinH, 2025-04-25:
In our own implementation, I changed that log message to FINE, as I found it quite strange and annoying as well. You get a warning for a resource not being a URL, but then afterwards it will still load it cause its actually a file...