ControlSystemStudio/phoebus #3382: Release built archives for 4.7.4

DenSinH, 2025-04-24:
Hello all,

We do an incremental build in our CI/CD based on the released phoebus zip files for version 4.7.3, we would like to migrate to 4.7.4, but there is no built archive for this version in the releases. Could this be added?

Another brief, kind of related question: is the release of 5.0.0 intended, it is not linked to the master branch it seems.

shroffk, 2025-04-24:
We will add the release Phoebus zip files. @georgweiss can you help with the mac ones ( the cross compiling has had issues in the past )

The master branch is protected... so the maven release cannot push a tag on the master branch as part of the `release:prepare`
I used a release branch to solve this limitation... does the tag not being on master introduce issues?

DenSinH, 2025-04-24:
The tag does not necessarily cause big issues, I was just wondering if it was by accident, or if there is an actual difference between 4.7.4 and 5.0.0! For some reason I can't cherry pick those commits or merge the v5.0.0 tag though, presumably since git doesn't fetch those as they are not linked to a branch (?)

kasemir, 2025-04-24:
> does the tag not being on master introduce issues?

Makes it harder to see what's what.
On https://github.com/ControlSystemStudio/phoebus/releases, v5.0.0 is listed with commit 3598ddb.
When I look in the list of recent commits, https://github.com/ControlSystemStudio/phoebus/commits/master/, there is no 3598ddb in there.

georgweiss, 2025-04-24:
@shroffk, where do I upload? https://github.com/ControlSystemStudio/phoebus/releases/tag/v5.0.0?

georgweiss, 2025-04-24:
Done.

closed 2025-08-21 by shroffk