ControlSystemStudio/phoebus #3397: Optimize widget info dialog to show archived status

shroffk, 2025-05-13:
There are a few improvements suggested in #3387 which should be incorporated in the info dialog.

georgweiss, 2025-05-14:
@shroffk, can you please clarify?

shroffk, 2025-05-14:
I am thinking primarily of finding a way to batch the queries.
While AA supports this... I am not sure how to do this via the archive:// datasource

we can also have the datasource return more interesting connection information which could make the column more useful... in addition to yes/no we can have last timestamp, currently connected, etc... again I am not sure how to do all of this but we can start with a list of additional meta data that would make this flag more useful.