ControlSystemStudio/phoebus #3398: Improve support for centrally managed mementos

eddybl, 2025-05-15:
### The Issue

Currently, memenots exist in the user space at one location. There is no simple way to provide centrally **managed** mementos **in addition** to user-controlled/personal mementos.

### Workarounds

There are some options to work around this issue. For example #2435 discusses the option of copying central mementos into the user space directory.

But the problem is that a simple file copy is not the same as a managed file. For example over time, if centrally managed memenots evolve (change name or get removed) they still exist in the user space.

### Possible Solution

I think a cleaner option would be to add second (optional) memento look-up location, which can be configured via the settings. With this, there would be no need to mess with the user space mementos. This second location could then be:

- just an additonal manually managed folder somewhere else
- a folder fully managed by salt/ansible
- a folder fully managed by launcher scripts
- a remote location?

### Implications for the GUI

Phoebus then would need to process both locations and in some form "merge" the mementos for the user to select from. The following options come to mind for the navigation menu;

1. do nothing extra, just present memenots from both locations in the same list
2. add a horizontal line seperator to distingiush the two locations
3. add another navigation level (right arrow which opens another menu) either just for the second location or both

To keep the impact low for people who do not want to use that, this feature could be behind a feature flag (probably not relevant for option 1)

smarsching, 2025-05-16:
Personally, I like options 1 and 2 best. I think that 3 would add an unnecessary layer of indirection, making the UI less user friendly.

When implementing option 1, I would choose the following approach: If there is a file with the same name in both the user location and the shared location, the one from the user location takes precedence. This way, the files in the shared location would act as „defaults“ that users can override if they like.

I suggest introducing a new preference `shared_mementos_directory` to the `org.phoebus.framework.workbench` preferences. This preference would be empty by default, thus disabling this feature completely and reverting to the current behavior. If this preference was set, the additional files would be loaded from the directory that is specified through the new preference.

smarsching, 2025-09-03:
I think that with the new `org.phoebus.ui/layout_dir` and `org.phoebus.ui/save_layout_in_layout_dir` preferences, we have all that we need.

The existing code in `PhoebusApplication.createLoadLayoutsMenu` of Phoebus 5.0.2 already loads the mementos in `layouts_dir` in addition to the ones in `~/.phobus`:

```java
            if (Preferences.layout_dir != null && !Preferences.layout_dir.isBlank()) {
                final File layoutDir = new File(Preferences.layout_dir);
                if (layoutDir.exists()) {
                    final File[] systemLayoutFiles = layoutDir.listFiles();
                    if (systemLayoutFiles != null) {
                        Arrays.stream(systemLayoutFiles).forEach(file -> {
                            if (!layoutFiles.containsKey(file.getName()) && file.getName().endsWith(".memento")) {
                                layoutFiles.put(file.getName(), file);
                            }
                        });
                    }
                }
            }
```

And when `save_layout_in_layout_dir` is `false` (which I think is the default), mementos saved by the user will still go into `~/.phoebus`, so users can keep their custom mementos in parallel to the provided ones.

So, it only is a matter of having the launcher prepare a directory with the memento files for the desired layouts.