ControlSystemStudio/phoebus #3436: Screen window sizing is not respected in JavaFX 21

rjwills28, 2025-06-18:
I have noticed a problem when using the command line to open a screen in a new window which specifies the **size** of that screen.

The command I use is:
``` 
phoebus.sh -resource  file:/home/xxx/bob/standalone1.bob?target=window@600x600+800+150
```

I would expect the size to be 600x600 but instead it always opens much smaller. No matter what value I specify for the screen size, this is what I get:

[image]

Interestingly I do _not_ see this issue in Phoebus v4.7.3. I have pinned it down to an issue in JavaFX that was not there in JavaFX 19 but is in JavaFX 21. It also depends on the gnome version being used - I have tested on Ubuntu24.04 and Fedora41 - both using later versions of gnome. Yet, the problem is not there in any version of Phoebus on RedHat 7 (using a much older version of gnome).

Here you can find the **bug report**: https://bugs.openjdk.org/browse/JDK-8316423 and the associated fix in JavaFX: https://github.com/openjdk/jfx/pull/1249. You can see it is consistent with my finding that it affects later versions of gnome as the underlying change came from an update to mutter.

**FIX OPTIONS**:
- The above fix is available in JavaFX 22 but not 21, therefore one solution to fix this in Phoebus would be to update to JavaFX22 which still requires JDK17 (JavaFX 23 would required JDK21).
- Alternatively, if this isn't an option, a workaround is to set the stage height and width _after_ the stage has been created, i.e. after `stage.show()` at line 173: https://github.com/ControlSystemStudio/phoebus/blob/0332a7fc0f1a6bd98ea16f46d94f94a36a4fd73e/core/ui/src/main/java/org/phoebus/ui/docking/DockStage.java#L173 
add:
   ```
   stage.setWidth(geometry.width);
   stage.setHeight(geometry.height);
   ```

What do people think is the best way forward here?

rjwills28, 2025-06-18:
And you don't think it will be a problem that then to use the latest version of Phoebus we will require users to update to JDK25? I only ask as I'm not sure how quick people are to adopt the latest version.

georgweiss, 2025-06-18:
Thing is there are other JavaFX bugs that require (if I recall correctly) at least JDK 24, which is not LTS.